chipsalliance/rocket-chip #3545: It is needed to clarify dependent tools version

SpinEch0, 2023-12-14:
Current master version:74d65ca320cf5f296b39dc5218f6f099e08e2794
It is confusing that which mill and firtool to use.  In Readme,  I find that mill 0.10.9 is mentioned, but it is not a worked version.
Actually, mill 0.11.6 works well.
What is more, there is also something wrong with firtool 1.61.0.

michael-etzkorn, 2023-12-23:
Merges to master would ideally be accompanied by release notes. My own volunteer time has been limited to help with changelog and automation. 

A first step would be something in vein of https://github.com/chipsalliance/rocket-chip/issues/3231 which would update the changelog and mark a commit as a released version.

From a code management perspective, this would look something like the following using git flow:

- release tags - designated commit versions for users. tooling and verification policies are documented to these commits. 

master - a stable edge for power users comfortable following tooling changes on the fly

dev - rc developers / bleeding edge users.

SpinEch0, 2023-12-26:
What about to add a new target "init" in Makefile? Just like this:
`
init:
      add init tools... ...

rocketchip: init
     ... ...
`

jlong299, 2024-01-16:
The version of firtool seems to be 1.30.0 for current master branch.  Correct me if not.

lordspacehog, 2024-02-28:
The project uses nix for dependency management for the env, and is the generally supported way to run things. You should be able to look through flake.nix, overlay.nix, and the nix lock file to back out those versions, but it will generally be easier to use nix.

SpinEch0, 2024-02-29:
> The project uses nix for dependency management for the env, and is the generally supported way to run things. You should be able to look through flake.nix, overlay.nix, and the nix lock file to back out those versions, but it will generally be easier to use nix.

Thanks. It helps a lot.

closed 2024-03-01 by SpinEch0