chipsalliance/rocket-chip #3584: Error in ./build.sh setp of rocket-tools

MateusFauri, 2024-03-11:
**Type of issue**: bug report 

**Impact**: no functional change

I'm using fedora 39 and  the actual commit hash is `ca6dc52742914ab5f9b7fd1444fa0ffbae9aa631

The problem is when I run `./build.sh` (i tried too  `./build.sh --with-arch=rv64gc_zfencei `) , but still get the same error:

```
Installing project fsf-binutils-gdb
libtool: install: warning: remember to run `libtool --finish /nowhere'

Removing existing riscv-pk/build directory
Configuring project riscv-pk
Building project riscv-pk
../dummy_payload/dummy_entry.S: Assembler messages:
../dummy_payload/dummy_entry.S:23: Warning: unterminated string; newline inserted
../dummy_payload/dummy_entry.S:24: Warning: unterminated string; newline inserted
../dummy_payload/dummy_entry.S:25: Warning: unterminated string; newline inserted
../dummy_payload/dummy_entry.S:26: Warning: unterminated string; newline inserted
../dummy_payload/dummy_entry.S:27: Warning: unterminated string; newline inserted
../dummy_payload/dummy_entry.S:28: Warning: unterminated string; newline inserted
../dummy_payload/dummy_entry.S:29: Warning: unterminated string; newline inserted
~/Documentos/rocket-chip/rocket-tools/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: warning: dummy_payload has a LOAD segment with RWX permissions
../machine/flush_icache.c: Assembler messages:
../machine/flush_icache.c:4: Error: unrecognized opcode `fence.i', extension `zifencei' required
gmake: *** [Makefile:336: flush_icache.o] Error 1
```

I can't found a solution for that in others issues.

Thanks!

lordspacehog, 2024-03-20:
@MateusFauri This project has moved to using nix for it's supported development environment. Im not sure how up-to-date the rocket-tools stuff is at this moment. If you don't want to use nix, i've had good luck with https://github.com/riscv-collab/riscv-gnu-toolchain. specifically you need to make sure you build your gcc with both the zicsr and zifencei extensions. They've also added the multilib generator if you're trying to develop against multiple arch types.

michael-etzkorn, 2024-07-24:
This build issue pops up from time to time over the years and the best way to nip it in the bud (IMO) would be to remove references to rocket-tools from the readme and [the hash file](https://github.com/chipsalliance/rocket-chip/blob/master/riscv-tools.hash). It's de-facto deprecated and has been a sore point of entry for people looking to try out RC.

michael-etzkorn, 2024-07-24:
You ran `./build.sh --with-arch=rv64gc_zfencei`

Did you try `--with-arch=rv64gc_zifencei` (note the `i` between `z` and `f`). There's an open issue https://github.com/chipsalliance/rocket-tools/issues/85 about this.