chipsalliance/rocket-chip #3601: Generation of  verilog file for rocket chip

srishti-sr, 2024-03-23:
I am a beginner working with rocket chip generator for my project but I am not able to generate the verilog file for it and facing this error. I am using an apple M1 chip macOS. I started with cloning the repository,installing the rocket tools for Mac (https://github.com/riscv-software-src/homebrew-riscv), jdk 17.0.10, sbt 1.9.9, Mill 0.11.7 and tried to generate the verilog. Are there some other steps involved?

srishtisharma@Srishtis-MacBook-Pro ~ % cd rocket-chip
srishtisharma@Srishtis-MacBook-Pro rocket-chip % make verilog
cd /Users/srishtisharma/rocket-chip && mill emulator[freechips.rocketchip.system.TestHarness,freechips.rocketchip.system.DefaultConfig].mfccompiler.compile
[172/172] emulator[freechips.rocketchip.system.TestHarness,freechips.rocketchip.system.DefaultConfig].mfccompiler.compile 
1 targets failed
emulator[freechips.rocketchip.system.TestHarness,freechips.rocketchip.system.DefaultConfig].mfccompiler.compile java.io.IOException: Cannot run program "firtool" (in directory "/Users/srishtisharma/rocket-chip/out/emulator/freechips.rocketchip.system.TestHarness/freechips.rocketchip.system.DefaultConfig/mfccompiler/compile.dest"): error=2, No such file or directory
    java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1170)
    java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1089)
    os.proc.proc$lzycompute$1(ProcessOps.scala:123)
    os.proc.proc$1(ProcessOps.scala:122)
    os.proc.spawn(ProcessOps.scala:129)
    os.proc.call(ProcessOps.scala:87)
    millbuild.build$Emulator$mfccompiler$.$anonfun$compile$3(build.sc:151)
java.io.IOException: error=2, No such file or directory
    java.base/java.lang.ProcessImpl.forkAndExec(Native Method)
    java.base/java.lang.ProcessImpl.<init>(ProcessImpl.java:295)
    java.base/java.lang.ProcessImpl.start(ProcessImpl.java:225)
    java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1126)
    java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1089)
    os.proc.proc$lzycompute$1(ProcessOps.scala:123)
    os.proc.proc$1(ProcessOps.scala:122)
    os.proc.spawn(ProcessOps.scala:129)
    os.proc.call(ProcessOps.scala:87)
    millbuild.build$Emulator$mfccompiler$.$anonfun$compile$3(build.sc:151)
make: *** [verilog] Error 1
srishtisharma@Srishtis-MacBook-Pro rocket-chip %

lordspacehog, 2024-03-25:
You need to have firtool installed since that's what's doing the verilog generation. The project has moved to using nix for it's dependencies/environment management so you should try getting that set up.

srishti-sr, 2024-03-25:
I have set up nix in my system but how can I install firtool

cap-oglu, 2024-04-03:
run `nix develop`, it will probably download and build tools which is necessary and includes firtool, if you want to download firtool binaries explicitly, download it from circt releases.

tzwaenn, 2024-04-04:
I have similar troubles - got the latest firtool binaries from circt, and getting an error because option -dedup does not exist there, just -no-dedup. Removing -dedup from build.sc allows
make verilog
to not error out, but the next one form the README
make verilog CONFIG=DefaultSmallConfig
still errors out with
[build.sc] [47/53] zincReportCachedProblems 
Cannot resolve emulator[freechips.rocketchip.system.TestHarness,DefaultSmallConfig].mfccompiler.compile. Try `mill resolve emulator._` or `mill resolve __.compile` to see what's available.
Would be great if the README was more up to date (firtool is not mentioned at all).

cap-oglu, 2024-04-05:
`make verilog CONFIG = freechips.rocketchip.system.DefaultSmallConfig` would not error out. Because when you write `make verilog CONFIG=DefaultSmallConfig` it looks for `[freechips.rocketchip.system.TestHarness,DefaultSmallConfig]` but it should look for `[freechips.rocketchip.system.TestHarness,freechips.rocketchip.system.DefaultSmallConfig]`

srishti-sr, 2024-04-13:
installing firtool helped thank you.

closed 2024-04-13 by srishti-sr

Amagicman, 2024-06-02:
> nix

how to cofigre nix

Amagicman, 2024-06-02:
> installing firtool helped thank you.

I have similar troubles ,how to installing firtool

jerryz123, 2024-06-02:
Beginners to rocket-chip should refer to chipyard for an integrated development environment https://github.com/ucb-bar/chipyard

Amagicman, 2024-06-04:
I followed this tutorial =》https://github.com/ucb-bar/chipyard =》 ./build-setup.sh riscv-tools # or esp-tools

Completed the following steps：
-s 1  initializing Conda environment

-s 2  initializing Chipyard submodules

-s 3  initializing toolchain collateral (Spike, PK, tests, libgloss)

-s 4  initializing ctags

-s 5  pre-compiling Chipyard Scala sources

when I make verilog in (base) hang@vm:~/work/lx002/chipyard/generators/rocket-chip$ make verilog

=>I have similar troubles too,How can I solve this problem

(base) hang@vm:~/work/lx002/chipyard/generators/rocket-chip$ make verilog
cd /home/hang/work/lx002/chipyard/generators/rocket-chip && mill emulator[freechips.rocketchip.system.TestHarness,freechips.rocketchip.system.DefaultConfig].mfccompiler.compile

.......
> Beginners to rocket-chip should refer to chipyard for an integrated development environment https://github.com/ucb-bar/chipyard

[216/219] emulator[freechips.rocketchip.system.TestHarness,freechips.rocketchip.system.DefaultConfig].generator.elaborate 
<stdin>:52.18-57.5: Warning (simple_bus_reg): /soc/cbus_clock: missing or empty reg/ranges property
<stdin>:75.28-78.5: Warning (simple_bus_reg): /soc/external-interrupts: missing or empty reg/ranges property
<stdin>:95.18-100.5: Warning (simple_bus_reg): /soc/pbus_clock: missing or empty reg/ranges property
<stdin>:36.29-40.6: Warning (interrupt_provider): /cpus/cpu@0/interrupt-controller: Missing #address-cells in interrupt provider
[219/219] emulator[freechips.rocketchip.system.TestHarness,freechips.rocketchip.system.DefaultConfig].mfccompiler.compile 
1 targets failed
emulator[freechips.rocketchip.system.TestHarness,freechips.rocketchip.system.DefaultConfig].mfccompiler.compile java.io.IOException: Cannot run program "firtool" (in directory "/home/hang/work/lx002/chipyard/generators/rocket-chip/out/emulator/freechips.rocketchip.system.TestHarness/freechips.rocketchip.system.DefaultConfig/mfccompiler/compile.dest"): error=2, 没有那个文件或目录
    java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1128)
    java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1071)
    os.proc.proc$lzycompute$1(ProcessOps.scala:123)
    os.proc.proc$1(ProcessOps.scala:122)
    os.proc.spawn(ProcessOps.scala:129)
    os.proc.call(ProcessOps.scala:87)
    millbuild.build$Emulator$mfccompiler$.$anonfun$compile$3(build.sc:170)
java.io.IOException: error=2, 没有那个文件或目录
    java.base/java.lang.ProcessImpl.forkAndExec(Native Method)
    java.base/java.lang.ProcessImpl.<init>(ProcessImpl.java:340)
    java.base/java.lang.ProcessImpl.start(ProcessImpl.java:271)
    java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1107)
    java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1071)
    os.proc.proc$lzycompute$1(ProcessOps.scala:123)
    os.proc.proc$1(ProcessOps.scala:122)
    os.proc.spawn(ProcessOps.scala:129)
    os.proc.call(ProcessOps.scala:87)
    millbuild.build$Emulator$mfccompiler$.$anonfun$compile$3(build.sc:170)
make: *** [Makefile:11：verilog] 错误 1

Amagicman, 2024-06-04:
> Beginners to rocket-chip should refer to chipyard for an integrated development environment https://github.com/ucb-bar/chipyard

I have similar troubles ,how to installing firtool/circt from nix,how to input commond

Amagicman, 2024-06-04:
> installing firtool helped thank you.

I have similar troubles ,how to installing firtool/circt from nix,how to input commond,thank you very much