chipsalliance/rocket-chip #3604: [Proposal] Refactor the remaining helpers out of the diplomacy module in rocket-chip.

lordspacehog, 2024-03-28:
**Type of issue**: other enhancement

**Impact**: API modification

**Development Phase**:  proposal

refactor freechips.rocketchip.diplomacy to move remaining old diplomacy helpers out to other modules or to the utils module.

lordspacehog, 2024-03-28:
@jerryrzhao @sequencer it seems like most of the remaining items in the diplomacy module dont really depend on diplomacy and are just helpers or utils. Any thoughts on moving these out to different areas so we can fully deprecate the diplomacy module in rocket-chip at some point?

jerryz123, 2024-03-28:
Do we need to deprecate it? I think for some things.. it might make sense to have a `freechips.rocketchip.diplomacy` that provides implementations of APIs from `org.chipsalliance.diplomacy`

Removing the alias could be done  removing `freechips.rocketchip.diplomacy`

lordspacehog, 2024-03-28:
I think short term we'd still want to alias these objects to preserve functionality. It's mostly that the stuff that's left isn't generally diplomacy specific. it's a mix of things, like clock crossings (has a note that it should move to PRCI which seems reasonable), the "Resource" stuff, DTS builder, and some tools for things like address and id ranges that don't depend on anything in the broken out diplomacy. They're also not direct implementations of anything specifically diplomacy related.

mostly that the cleanup will help keep things a little more usefully organized.

sequencer, 2024-03-28:
DTS is harder then what I expected: at least blocked by:
https://github.com/chipsalliance/chisel/issues/3939
This is a tricky issue for D/I integrations...

I personally would like to keep this refactoring slow, until we really achieve it.
I'm experimentally using it for T1, I'll take a look at it until I get thing right.

lordspacehog, 2024-03-29:
yeah, i dont think we should remove code, or really change functionality right now. I mostly just want to do a little organization to clarify what it's actually doing.