chipsalliance/rocket-chip #3607: Bus error unit cannot be added, fails diplomacy

Kevin99214, 2024-04-03:
**Type of issue**: bug report 

**Impact**:  unknown

Currently working on Chipyard v1.11 release. I'm trying to enable  the bus error unit to detect ECC errors. 
I wrote a simple config snippet that I added to the base RocketConfig

```
class WithBEU(addr: BigInt) extends Config((site, here, up) => {
  case TilesLocated(InSubsystem) => up(TilesLocated(InSubsystem), site) map {
    case tp: RocketTileAttachParams => tp.copy(tileParams = tp.tileParams.copy(beuAddr = Some(addr)))
    case t => t
  }
})
```

```
class RocketConfig extends Config(
  new freechips.rocketchip.subsystem.WithBEU(addr = 0xC8010000L) ++ 
  new freechips.rocketchip.subsystem.WithNBigCores(1) ++         // single rocket-core
  new chipyard.config.AbstractConfig)
```

When I try to generate the RTL for this, it runs into a diplomacy issue
```
Exception in thread "main" java.lang.reflect.InvocationTargetException
	at ... ()
	at chipyard.stage.phases.PreElaboration.$anonfun$transform$1(PreElaboration.scala:35)
	at ... ()
	at ... (Stack trace trimmed to user code only. Rerun with --full-stacktrace to see the full stack trace)
Caused by: java.lang.IllegalArgumentException: requirement failed: Diplomacy has detected a problem with your graph:
The following node appears left of a :*= 1 times and right of a :=* 1 times, at most once is allowed.
ephemeral system.tileToPlicNodes node:
parents: system/chiptop0
locator:  (generators/chipyard/src/main/scala/ChipTop.scala:27:35)

1 inward nodes bound: [star-intsink.node]
1 outward nodes bound: [star-plic.intnode]

	at scala.Predef$.require(Predef.scala:337)
	at freechips.rocketchip.diplomacy.MixedAdapterNode.resolveStar(Nodes.scala:1446)
	at freechips.rocketchip.diplomacy.MixedNode.liftedTree1$1(Nodes.scala:1061)
	at freechips.rocketchip.diplomacy.MixedNode.x$24$lzycompute(Nodes.scala:1032)
	at freechips.rocketchip.diplomacy.MixedNode.x$24(Nodes.scala:1031)
	at freechips.rocketchip.diplomacy.MixedNode.oStar$lzycompute(Nodes.scala:1031)
	at freechips.rocketchip.diplomacy.MixedNode.oStar(Nodes.scala:1031)
	at freechips.rocketchip.diplomacy.MixedNode.$anonfun$x$24$6(Nodes.scala:1058)
	at freechips.rocketchip.diplomacy.MixedNode.$anonfun$x$24$6$adapted(Nodes.scala:1055)
	at scala.collection.immutable.List.map(List.scala:250)
	at scala.collection.immutable.List.map(List.scala:79)
	at freechips.rocketchip.diplomacy.MixedNode.liftedTree1$1(Nodes.scala:1055)
	at freechips.rocketchip.diplomacy.MixedNode.x$24$lzycompute(Nodes.scala:1032)
	at freechips.rocketchip.diplomacy.MixedNode.x$24(Nodes.scala:1031)
	at freechips.rocketchip.diplomacy.MixedNode.iPortMapping$lzycompute(Nodes.scala:1031)
	at freechips.rocketchip.diplomacy.MixedNode.iPortMapping(Nodes.scala:1031)
	at freechips.rocketchip.diplomacy.MixedNode.$anonfun$oDirectPorts$1(Nodes.scala:1094)
	at scala.collection.immutable.List.flatMap(List.scala:293)
	at scala.collection.immutable.List.flatMap(List.scala:79)
	at freechips.rocketchip.diplomacy.MixedNode.oDirectPorts$lzycompute(Nodes.scala:1092)
	at freechips.rocketchip.diplomacy.MixedNode.oDirectPorts(Nodes.scala:1092)
	at freechips.rocketchip.diplomacy.MixedNode.oPorts$lzycompute(Nodes.scala:1137)
	at freechips.rocketchip.diplomacy.MixedNode.oPorts(Nodes.scala:1137)
	at freechips.rocketchip.diplomacy.MixedNode.outputs(Nodes.scala:1311)
	at freechips.rocketchip.interrupts.IntXbar$$anon$1.circuitIdentity(Xbar.scala:18)
	at freechips.rocketchip.diplomacy.MixedNode.instantiate(Nodes.scala:1280)
	at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$16(LazyModule.scala:343)
	at scala.collection.immutable.List.flatMap(List.scala:293)
	at freechips.rocketchip.diplomacy.LazyModuleImpLike.instantiate(LazyModule.scala:343)
	at freechips.rocketchip.diplomacy.LazyModuleImpLike.instantiate$(LazyModule.scala:307)
	at freechips.rocketchip.diplomacy.LazyRawModuleImp.instantiate(LazyModule.scala:406)
[... 143 lines cut ...]
	at firrtl.options.DependencyManager.transform$(DependencyManager.scala:255)
	at firrtl.options.PhaseManager.transform(DependencyManager.scala:443)
	at chisel3.stage.ChiselStage.run(ChiselStage.scala:46)
	at firrtl.options.Stage$$anon$1.transform(Stage.scala:43)
	at firrtl.options.Stage$$anon$1.transform(Stage.scala:43)
	at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:38)
	at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:15)
	at firrtl.options.Translator.transform(Phase.scala:248)
	at firrtl.options.Translator.transform$(Phase.scala:248)
	at firrtl.options.phases.DeletedWrapper.transform(DeletedWrapper.scala:15)
	at firrtl.options.Stage.$anonfun$transform$5(Stage.scala:47)
	at scala.collection.LinearSeqOps.foldLeft(LinearSeq.scala:183)
	at scala.collection.LinearSeqOps.foldLeft$(LinearSeq.scala:179)
	at scala.collection.immutable.List.foldLeft(List.scala:79)
	at firrtl.options.Stage.$anonfun$transform$3(Stage.scala:47)
	at logger.Logger$.$anonfun$makeScope$2(Logger.scala:137)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:59)
	at logger.Logger$.makeScope(Logger.scala:135)
	at firrtl.options.Stage.transform(Stage.scala:47)
	at firrtl.options.Stage.execute(Stage.scala:58)
	at firrtl.options.StageMain.main(Stage.scala:71)
	at chipyard.Generator.main(Generator.scala)
```

It looks like something is happening when it tries to connect interrupts to the PLIC. Is bus error unit no longer supported? Not sure where to go from this so any advice on how to debug this would be great.

jerryz123, 2024-04-03:
Try patching in the fix in #3608. This will get merged and propagate into chipyard soon.

Kevin99214, 2024-04-03:
Thanks. It works now!

closed 2024-04-03 by Kevin99214