chipsalliance/rocket-chip #3623: Error when building the r

iagrigorov, 2024-04-22:


iagrigorov, 2024-04-22:
Hello everyone!
When I am trying to build the project I get the following error:
```
1 targets failed
emulator[freechips.rocketchip.system.TestHarness,freechips.rocketchip.system.DefaultConfig].generator.elaborate os.SubprocessException: Result of /usr/lib/jvm/java-11-openjdk-amd64/bin/java…: 1
Interrupt map (2 harts 2 interrupts):
  [1, 2] => ldut

    os.proc.call(ProcessOps.scala:89)
    millbuild.build$Emulator$generator$.$anonfun$elaborate$2(build.sc:126)
    mill.define.Task$TraverseCtx.evaluate(Task.scala:71)
make: *** [Makefile:11: verilog] Error 1
```
What can cause this problem?

jerryz123, 2024-05-15:
I can only assume this is due to some issue with your system Java.
For an out-of-the-box environment that builds and simulates rocket-chip-based SoCs, you can try chipyard. https://github.com/ucb-bar/chipyard

Chipyard  is a bit more user-friendly than the base rocket-chip repo.

SpinEch0, 2024-06-05:
Did you install dtc(device tree compiler) ?
```
Caused by: java.lang.IllegalArgumentException: requirement failed: Failed to run dtc; is it in your path?
	at scala.Predef$.require(Predef.scala:337)
	at freechips.rocketchip.diplomacy.DTB$.apply(DeviceTree.scala:136)

```

iagrigorov, 2024-07-15:
> Did you install dtc(device tree compiler) ?
> 
> ```
> Caused by: java.lang.IllegalArgumentException: requirement failed: Failed to run dtc; is it in your path?
> 	at scala.Predef$.require(Predef.scala:337)
> 	at freechips.rocketchip.diplomacy.DTB$.apply(DeviceTree.scala:136)
> ```
Thank you for your reply.
Yes, I did. My version is 1.5.1-1. Maybe wrong version.

> I can only assume this is due to some issue with your system Java. For an out-of-the-box environment that builds and simulates rocket-chip-based SoCs, you can try chipyard. https://github.com/ucb-bar/chipyard
> 
> Chipyard is a bit more user-friendly than the base rocket-chip repo.

Ok, I see. Thank you!

xionyuno123, 2024-10-07:
I have the same problem. Have you solved it now?