chipsalliance/rocket-chip #3630: Discrepency between simulation with and without traces

bantierr, 2024-05-14:
**Type of issue**: bug report

**Impact**: unknown

**Other information**
 
I have created a dedicated repository for the bug at: [https://github.com/bantierr/rocket_issue.git](https://github.com/bantierr/rocket_issue.git)

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
I am simulating an automatically generated program, and I am experiencing outputs mismatches between a simulation with traces and without traces on verilator. Please, find all the relevant files in the repository attached above. It seems like this behavior also happens on modelsim.

**What is the current behavior?**

When simulating the program without traces, the program behaves correctly. The program executes an ECALL instruction from Supervisor mode, and then reads mcause, which holds the expected value for Environment call from S-mode (0x9). When simulating the program with traces, mcause holds the value for Supervisor external interrupt (0x8000000000000009).

**What is the expected behavior?**

I expect the simulation to have the same behavior with and without traces. 

**Please tell us about your environment:**

rocket-chip hash: d03aec2643b152e81468ee36f3bfd45872c14d6a
verilator hash: 5b8941d7170b0c11eff46703514b064becd3a648
OS: Ubuntu 22.04.3 LTS

jerryz123, 2024-05-15:
I was unable to reproduce the failing issue across Verilator and VCS, with and without traces (waveforms, I assume).
All the simulated tests returned 0x9 for mcause in the instruction at 0x8002f810.

I am using the latest chipyard to run the simulation, with Verilator 5.010 and a host gcc-10.5.0.

In the past, I've noticed divergences between verilator and verilator-debug simulations disappear after switching to a newer Verilator version. I can only assume this is due to some bug in Verilator.

bantierr, 2024-05-16:
Thank you so much for you answer!
I will double check my gcc version and get the latest verilator version. 

Just out of curiosity, did you use the optimization flags from the repo I linked ? I noticed some differences in the simulation when I use different optimizations on my verilator version.

Best,
Quentin

closed 2024-05-16 by bantierr