chipsalliance/rocket-chip #3646: class RocketChip$macro$3 needs to be abstract.

Amagicman, 2024-06-12:
how to deal with it??
(base) go@go~/work/l002/chipyard/generators/rocket-chip$ make verilog
cd /home/go/work/lx002/chipyard/generators/rocket-chip && mill emulator[freechips.rocketchip.system.TestHarness,freechips.rocketchip.system.DefaultConfig].mfccompiler.compile
[build.sc] [49/53] compile
[info] compiling 4 Scala sources to /home/go/work/lx002/chipyard/generators/rocket-chip/out/mill-build/compile.dest/classes ...
[error] /home/go/work/lx002/chipyard/generators/rocket-chip/build.sc:65:72: class RocketChip$macro$3 needs to be abstract.
[error] Missing implementation for member of trait RocketChipModule:
[error]   def diplomacyModule: mill.scalalib.ScalaModule = ???
[error] object rocketchip extends Cross[RocketChip](v.chiselCrossVersions.keys.toSeq)
[error]                                                                        ^
[warn] /home/go/work/lx002/chipyard/generators/rocket-chip/build.sc:205:155: method any2stringadd in object Predef is deprecated (since 2.13.0): Implicit injection of + is deprecated. Convert to String to call +
[warn]          |"$${CMAKE_CXX_FLAGS} -DVERILATOR -DTEST_HARNESS=VTestHarness -include VTestHarness.h -include verilator.h -include ${generator.elaborate().path / config + ".plusArgs"}")
[warn]
                                         ^
[warn] one warning found
[error] one error found
1 targets failed
compile Compilation failed
make: *** [Makefile:11: verilog] Error 1

sequencer, 2024-06-12:
Please use nix to setup the building environment.

closed 2024-06-12 by sequencer

Amagicman, 2024-06-12:
Thank you very much. After resetting the original environment using nix, it can now be compiled,
[note] A new release of pip available: 22.3.1->24.0
[note] To update, run: python3.10 -m pip install --upgrade pip

yulong-lan, 2024-08-05:
Hi @Amagicman Do you mind letting me know how you did it exactly using nix?