chipsalliance/rocket-chip #3647: Bus Error Unit cannot find implicit clock when RocketTile crossingType is set to RationalCrossing

Kevin99214, 2024-06-12:
**Type of issue**: bug report

**Impact**: unknown

Follow up from #3607. 

I now want to change my design to have 2 main clock domains, a core clock and an uncore clock, with core clock being two times the frequency of uncore clock.

So I created this snippet
```
class WithRocketTileCDC(crossingType: ClockCrossingType = SynchronousCrossing()) extends Config((site, here, up) => {
  case TilesLocated(InSubsystem) => up(TilesLocated(InSubsystem), site) map {
    case tp: RocketTileAttachParams => tp.copy(crossingParams = tp.crossingParams.copy(
      crossingType = crossingType
    ))
    case other => other
  }
})
```

And I add it to the RocketConfig
```
class RocketConfig extends Config(
  new freechips.rocketchip.subsystem.WithRocketTileCDC(
    crossingType = RationalCrossing(direction = FastToSlow)) ++
  new freechips.rocketchip.subsystem.WithBEU(addr = 0xC8010000L) ++ 
  new freechips.rocketchip.subsystem.WithNBigCores(1) ++         // single rocket-core
  new chipyard.config.AbstractConfig)
```

When I generated the RTL, it errors out
```
Exception in thread "main" java.lang.reflect.InvocationTargetException
	at ... ()
	at chipyard.stage.phases.PreElaboration.$anonfun$transform$1(PreElaboration.scala:35)
	at ... ()
	at ... (Stack trace trimmed to user code only. Rerun with --full-stacktrace to see the full stack trace)
Caused by: chisel3.package$ChiselException: Error: No implicit clock.
	at chisel3.internal.throwException$.apply(Error.scala:93)
	at chisel3.internal.Builder$.$anonfun$forcedClock$1(Builder.scala:739)
	at scala.Option.getOrElse(Option.scala:201)
	at chisel3.internal.Builder$.forcedClock(Builder.scala:739)
	at chisel3.Module.$anonfun$initializeInParent$2(Module.scala:197)
	at scala.Option.getOrElse(Option.scala:201)
	at chisel3.Module.$anonfun$initializeInParent$1(Module.scala:197)
	at chisel3.Data.$anonfun$$colon$eq$1(Data.scala:720)
	at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.scala:18)
	at chisel3.experimental.prefix$.apply(prefix.scala:33)
	at chisel3.Data.$colon$eq(Data.scala:720)
	at chisel3.Module.initializeInParent(Module.scala:197)
	at chisel3.Module$.do_apply(Module.scala:83)
	at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$14(LazyModule.scala:336)
	at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:33)
	at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$13(LazyModule.scala:336)
	at scala.Option.getOrElse(Option.scala:201)
	at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$1(LazyModule.scala:334)
	at scala.collection.immutable.List.flatMap(List.scala:293)
	at freechips.rocketchip.diplomacy.LazyModuleImpLike.instantiate(LazyModule.scala:310)
	at freechips.rocketchip.diplomacy.LazyModuleImpLike.instantiate$(LazyModule.scala:307)
	at freechips.rocketchip.diplomacy.LazyRawModuleImp.instantiate(LazyModule.scala:406)
	at freechips.rocketchip.diplomacy.LazyRawModuleImp.$anonfun$x$23$1(LazyModule.scala:423)
	at chisel3.internal.plugin.package$.autoNameRecursivelyProduct(package.scala:48)
	at freechips.rocketchip.diplomacy.LazyRawModuleImp.<init>(LazyModule.scala:420)
	at freechips.rocketchip.subsystem.BareSubsystemModuleImp.<init>(BaseSubsystem.scala:29)
	at freechips.rocketchip.subsystem.BaseSubsystemModuleImp.<init>(BaseSubsystem.scala:135)
	at chipyard.TTChipyardSubsystemModuleImp.<init>(TTChipyardSubsystem.scala:90)
	at chipyard.TTChipyardSystemModule.<init>(TTChipyardSystem.scala:72)
	at chipyard.DigitalTopModule.<init>(DigitalTop.scala:46)
	at chipyard.DigitalTop.$anonfun$module$1(DigitalTop.scala:43)
	at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:33)
[... 91 lines cut ...]
	at firrtl.options.DependencyManager.transform$(DependencyManager.scala:255)
	at firrtl.options.PhaseManager.transform(DependencyManager.scala:443)
	at chisel3.stage.ChiselStage.run(ChiselStage.scala:46)
	at firrtl.options.Stage$$anon$1.transform(Stage.scala:43)
	at firrtl.options.Stage$$anon$1.transform(Stage.scala:43)
	at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:38)
	at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:15)
	at firrtl.options.Translator.transform(Phase.scala:248)
	at firrtl.options.Translator.transform$(Phase.scala:248)
	at firrtl.options.phases.DeletedWrapper.transform(DeletedWrapper.scala:15)
	at firrtl.options.Stage.$anonfun$transform$5(Stage.scala:47)
	at scala.collection.LinearSeqOps.foldLeft(LinearSeq.scala:183)
	at scala.collection.LinearSeqOps.foldLeft$(LinearSeq.scala:179)
	at scala.collection.immutable.List.foldLeft(List.scala:79)
	at firrtl.options.Stage.$anonfun$transform$3(Stage.scala:47)
	at logger.Logger$.$anonfun$makeScope$2(Logger.scala:137)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:59)
	at logger.Logger$.makeScope(Logger.scala:135)
	at firrtl.options.Stage.transform(Stage.scala:47)
	at firrtl.options.Stage.execute(Stage.scala:58)
	at firrtl.options.StageMain.main(Stage.scala:71)
	at chipyard.Generator.main(Generator.scala)
```

With a different clock domain in the RocketTile, the bus error unit doesn't have implicit clock anymore? Any help with this is greatly appreciated

jerryz123, 2024-06-13:
Try #3648 . Missed this when removing the clock from BaseSubsystem, thanks for catching it.

If you have time, can you PR any config-fragments you find to be missing into this repo or chipyard? It would help with reproducing these kinds of errors.

Kevin99214, 2024-06-13:
Thanks, that has fixed it.

Yep I can do that, I'll make a PR with all the config-fragments I've been using soon.

PR #3648 (2024-06-13): Interrupts coming outwards from the Tile should cross into a toPlicDomain
This avoids accidental no-implicit-clock errors, now that BaseSubsystem has no clocks

Resolves #3647 

**Related issue**: 

**Type of change**: bug report | feature request | other enhancement

**Impact**: no functional change | API addition (no impact on existing code) | API modification

**Development Phase**: proposal |  implementation

**Release Notes**
closed 2024-06-13 by Kevin99214