chipsalliance/rocket-chip #3656: unable to build the C emulator due to missing verilog.h file

nhuynh368, 2024-07-03:
**Type of issue**: bug report

**Impact**: no functional change

**Other information**
 
Currently, when I try to build the C emulator, the Verilator version for the emulator failed to build due to a missing Verilog.h file. I'm guessing this is due to an incorrect bison installation, but bison was installed correctly on the PATH. Here is the build log: pastebin.com/ceKsZwrR

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

**What is the current behavior?**
Verilator installation failed mid parse.

**What is the expected behavior?**
Verilator installs correctly and the C emulator is built correctly.

**Please tell us about your environment:**

- version: v1.6
- OS: Ubuntu 24.04 LTS x86_64 GNOME 46.0

JackySu, 2025-02-25:
exact same error here, nothing I could help either :(

jerryz123, 2025-02-25:
Rocketchip should be used through Chipyard. This repo as it stands is intended to be used as a library for other frameworks like Chipyard.