chipsalliance/rocket-chip #3657: Make verilog Error

Krishnakumarmohanraj, 2024-07-24:
Hi Team,
 
  we installed rocket tools and chisel in machine and set the path, when we run the cmd "make verilog" in rocket-chip we are geeting attached error. Can you help me on this issue please
[image]

nhuynh368, 2024-07-24:
Same here. It seems like firtool isn't being built properly and I narrowed down the error portion to this specific piece of code in the build.sc file:

```
object mfccompiler extends Module {
    def compile = T {
      os.proc("firtool",
        generator.chirrtl().path,
        s"--annotation-file=${generator.chiselAnno().path}",
        "--disable-annotation-unknown",
        "-dedup",
        "-O=debug",
        "--split-verilog",
        "--preserve-values=named",
        "--output-annotation-file=mfc.anno.json",
        s"-o=${T.dest}"
      ).call(T.dest)
      PathRef(T.dest)
    }
```

yulong-lan, 2024-08-05:
so what is debup? I have firtool installed and when i run make verilog, the following error occurs:
`No mill version specified.
You should provide a version via '.mill-version' file or --mill-version option.
Using mill version 0.11.11
[189/189] emulator[freechips.rocketchip.system.TestHarness,freechips.rocketchip.system.DefaultConfig].mfccompiler.compile 
firtool: Unknown command line argument '-dedup'.  Try: 'firtool --help'
firtool: Did you mean '--no-dedup'?
1 targets failed
emulator[freechips.rocketchip.system.TestHarness,freechips.rocketchip.system.DefaultConfig].mfccompiler.compile os.SubprocessException: Result of firtool…: 1

    os.proc.call(ProcessOps.scala:95)
    millbuild.build$Emulator$mfccompiler$.$anonfun$compile$3(build.sc:151)
make: *** [Makefile:11: verilog] Error 1`

jerryz123, 2024-08-05:
To generate Verilog from rocket-chip, I recommend using Chipyard. The rocket-chip repo is intended to be used as a library by more complex projects.

Zou1c, 2024-08-10:
> so what is debup? I have firtool installed and when i run make verilog, the following error occurs:
>
> ```
> `No mill version specified.
> You should provide a version via '.mill-version' file or [...]

@yulong-lan
Maybe you can modify build.sc, about line 145 in object mfccompiler, change the parameter "-dedup" to "--no-dedup". I guess it's due to the update of firtool.(and my firtool version is 1.62.0)

Ashbin9400, 2025-04-23:
🔧 **firtool not found or fails with -dedup error**
It looks like you're running into a version mismatch with firtool. The current RocketChip build expects an older version that still supports the -dedup flag. Here's how to fix it:

✅ **Steps to Resolve**
_1. Check Current firtool Version_
Run:

bash

firtool --version
If the version is 1.113.0 or newer, that’s likely the issue — -dedup was removed in later versions.

**2. Download a Compatible Version**
_Recommended: firtool v1.51.0 (or v1.38.0)_

Go to: [CIRCT Releases](https://github.com/llvm/circt/releases?q=1.51.0&expanded=true)
Download: circt-linux-release-1.51.0.tar.gz

Then:

bash

cd ~/tools
tar -xvzf circt-linux-release-1.51.0.tar.gz

**3. Set Up PATH and LD_LIBRARY_PATH**
_Assuming it extracts to ~/tools/firtool-1.51.0-1-g..., do:_

bash

export PATH="$HOME/tools/firtool-1.51.0-1-g405b80517/bin:$PATH"
export LD_LIBRARY_PATH="$HOME/tools/circt/lib:$LD_LIBRARY_PATH"

Confirm:

bash

which firtool
firtool --version

Expected:

bash

/home/hdg/tools/firtool-1.51.0-1-g405b80517/bin/firtool
firtool version 1.51.0

**4. Rebuild Verilog**
_Go back to your RocketChip repo and run:_

bash
cd ~/HDG/rocket-chip
make verilog

This should now work without the -dedup error, and the Verilog will be generated in:

bash

rocket-chip/out/emulator/freechips.rocketchip.system.TestHarness/freechips.rocketchip.system.DefaultConfig/mfccompiler/compile.dest

⚠️ **If firtool Still Isn’t Found:**
_

[image]

_

bash
~/tools/firtool-1.51.0-1-g405b80517/bin/firtool --help
If you see missing library errors, make sure LD_LIBRARY_PATH points to where CIRCT’s .so files are.

Also, ensure no conflicting firtool versions are lingering in your PATH. Clean it up like this:

bash
export PATH="$HOME/tools/firtool-1.51.0-1-g405b80517/bin:$HOME/tools/circt/bin:$PATH"