chipsalliance/rocket-chip #3658: Java compilation error, prompting no byte 4096 alignment

Comet959, 2024-07-25:
Everything went wrong when I added support for hypervisor extensions to my configuration file.
This error was prompted[image]

My configuration file looks like this：

```
package freechips.rocketchip.uintr
  
import freechips.rocketchip.config._
import freechips.rocketchip.diplomacy._
import freechips.rocketchip.subsystem._
import freechips.rocketchip.system._
import freechips.rocketchip.tile._
import freechips.rocketchip.devices.tilelink._

class WithCustomBootROM(resetAddress: BigInt, bootImgPath: String)
  extends Config((_, _, up) => { case BootROMLocated(x) =>
    up(BootROMLocated(x)).map(_.copy(hang = resetAddress, contentFileName = bootImgPath))
  })

class WithUIPI extends Config((_, _, _) => {
  case BuildRoCC => Seq((p: Parameters) => {
    val module = LazyModule(new UIPI(OpcodeSet.custom3)(p))
    module
  })
})

class UintrConfig extends Config(
  new WithHypervisor ++
  new WithNBigCores(4) ++
    new WithNExtTopInterrupts(6) ++
    new WithTimebase((BigInt(10000000))) ++ // 10 MHz
    new WithDTS("freechips.rocketchip-unknown", Nil) ++
    new WithUIPI ++
    new WithCustomBootROM(0x10000, "../common/boot/bootrom/bootrom.img") ++
    new WithDefaultMemPort ++
    new WithDefaultMMIOPort ++
    new WithoutTLMonitors ++
    new WithCoherentBusTopology ++
    new BaseSubsystemConfig
)

class UintrTestConfig extends Config(new DefaultConfig ++ new WithUIPI)
```

The entire error message is：

[success] Total time: 71 s (01:11), completed Jul 26, 2024 4:18:29 AM
mkdir -p /home/jerry/bitstream/uintr-rocket-chip/rocket-chip/emulator/generated-src/
cd /home/jerry/bitstream/uintr-rocket-chip/rocket-chip && java -Xmx2G -Xss8M -cp /home/jerry/bitstream/uintr-rocket-chip/rocket-chip/rocketchip.jar freechips.rocketchip.uintr.Generator -td /home/jerry/bitstream/uintr-rocket-chip/rocket-chip/emulator/generated-src -T freechips.rocketchip.uintr.Top -C freechips.rocketchip.uintr.UintrConfig 
Init BootROM 65536 ../common/boot/bootrom/bootrom.img
Exception in thread "main" java.lang.reflect.InvocationTargetException
        at ... ()
        at freechips.rocketchip.stage.phases.PreElaboration.$anonfun$transform$1(PreElaboration.scala:36)
        at ... ()
        at ... (Stack trace trimmed to user code only. Rerun with --full-stacktrace to see the full stack trace)
Caused by: java.lang.IllegalArgumentException: requirement failed: All memory regions must be 4096-byte aligned
        at scala.Predef$.require(Predef.scala:281)
        at freechips.rocketchip.rocket.PTW.$anonfun$new$218(PTW.scala:378)
        at freechips.rocketchip.rocket.PTW.$anonfun$new$218$adapted(PTW.scala:375)
        at scala.collection.TraversableLike.$anonfun$map$1(TraversableLike.scala:286)
        at scala.collection.immutable.Range.foreach(Range.scala:158)
        at scala.collection.TraversableLike.map(TraversableLike.scala:286)
        at scala.collection.TraversableLike.map$(TraversableLike.scala:279)
        at scala.collection.AbstractTraversable.map(Traversable.scala:108)
        at freechips.rocketchip.rocket.PTW.$anonfun$new$217(PTW.scala:375)
        at chisel3.internal.prefix$.apply(prefix.scala:48)
        at freechips.rocketchip.rocket.PTW.$anonfun$new$216(PTW.scala:117)
        at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:33)
        at freechips.rocketchip.rocket.PTW.$anonfun$new$4(PTW.scala:117)
[... 140 lines cut ...]
        at freechips.rocketchip.system.RocketChipStage.run(RocketChipStageGenerator.scala:44)
        at firrtl.options.Stage$$anon$1.transform(Stage.scala:43)
        at firrtl.options.Stage$$anon$1.transform(Stage.scala:43)
        at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:38)
        at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:15)
        at firrtl.options.Translator.transform(Phase.scala:248)
        at firrtl.options.Translator.transform$(Phase.scala:248)
        at firrtl.options.phases.DeletedWrapper.transform(DeletedWrapper.scala:15)
        at firrtl.options.Stage.$anonfun$transform$5(Stage.scala:47)
        at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:126)
        at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:122)
        at scala.collection.immutable.List.foldLeft(List.scala:91)
        at firrtl.options.Stage.$anonfun$transform$3(Stage.scala:47)
        at logger.Logger$.$anonfun$makeScope$2(Logger.scala:137)
        at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
        at logger.Logger$.makeScope(Logger.scala:135)
        at firrtl.options.Stage.transform(Stage.scala:47)
        at firrtl.options.Stage.execute(Stage.scala:58)
        at firrtl.options.StageMain.main(Stage.scala:71)
        at freechips.rocketchip.uintr.Generator.main(Generator.scala)
make[1]: *** [Makefrag-verilator:13: /home/jerry/bitstream/uintr-rocket-chip/rocket-chip/emulator/generated-src/freechips.rocketchip.uintr.UintrConfig.fir] Error 1
make[1]: Leaving directory '/home/jerry/bitstream/uintr-rocket-chip/rocket-chip/emulator'
make: *** [../common/Makefrag:44: /home/jerry/bitstream/uintr-rocket-chip/rocket-chip/emulator/generated-src/freechips.rocketchip.uintr.UintrConfig.v] Error  2

How can i solve this?

Comet959, 2024-07-26:
I found the error come from here:

https://github.com/U-interrupt/rocket-chip/blob/2a66519752149804d9e0bad99326ee51687b70fa/src/main/scala/rocket/PTW.scala#L375C1-L383C4

do you have any ideas?

jerryz123, 2024-07-26:
The error message is descriptive:
```Caused by: java.lang.IllegalArgumentException: requirement failed: All memory regions must be 4096-byte aligned
at scala.Predef$.require(Predef.scala:281)```

Comet959, 2024-07-26:
> The error message is descriptive: `All memory regions must be 4096-byte aligned at scala.Predef$.require(Predef.scala:281)`

I have no idea to deal this exception.

Comet959, 2024-07-26:
This error only occurs when I turn on support for hypervisor extensions.

Jerryy959, 2024-08-21:
Check this issue to track my solution, this can be closed.
https://github.com/BOSC-Hvisor/hvisor-rocket-chip/issues/3

closed 2024-08-21 by Jerryy959