chipsalliance/rocket-chip #3662: ./build.sh not found and make verilog error

gituserdeepika, 2024-08-05:
I am trying to build the rocket-chip project by following the steps provided in https://github.com/chipsalliance/rocket-chip/blob/master/README.md#quick 
The git repo cloned successfully from https://github.com/ucb-bar/rocket-chip.git
Further while trying to install the necessary dependencies from https://github.com/chipsalliance/rocket-tools/blob/master/README.md
The 'Quickstart' mentions 3 steps as follows:
$ git submodule update --init --recursive
$ export RISCV=/path/to/install/riscv/toolchain
$ ./build.sh
But I couldn't find any build.sh in the rocket-chip directory. So I skipped this step.
And without this if I proceed to 'make verilog', I get the below error, Please clarify.

**Environment:**
WSL 2 on Windows 11
Ubuntu version - "Ubuntu 22.04.4 LTS (Jammy Jellyfish)"
git version 2.34.1
openjdk 11.0.24 2024-07-16
sbt version in this project: 1.4.9

**Error**
dgangana@Eva:~/rocketchip/rocket-chip$ make verilog
cd /home/dgangana/rocketchip/rocket-chip && mill emulator[freechips.rocketchip.system.TestHarness,freechips.rocketchip.system.DefaultConfig].mfccompiler.compile
Compiling /home/dgangana/rocketchip/rocket-chip/build.sc
/home/dgangana/rocketchip/rocket-chip/build.sc:588: not found: value Cross
trait JTAGDTMTest extends Cross.Module5[String, String, String, String, String] {
                          ^
/home/dgangana/rocketchip/rocket-chip/build.sc:462: not found: value Cross
trait ArchTest extends Cross.Module4[String, String, String, String] {
                       ^
/home/dgangana/rocketchip/rocket-chip/build.sc:434: not found: value Cross
trait RiscvTest extends Cross.Module4[String, String, String, String] {

jerryz123, 2024-08-05:
Standalone builds of rocket-chip are not well supported. Users looking to generate RTL from rocket-chip should use Chipyard.

gituserdeepika, 2024-08-05:
For building chipyard based rocket-chip, I followed the instructions from https://chipyard.readthedocs.io/en/latest/Chipyard-Basics/Initial-Repo-Setup.html#initial-repository-setup.

Installed Conda with the following steps:
curl -L -O "https://github.com/conda-forge/miniforge/releases/latest/download/Miniforge3-$(uname)-$(uname -m).sh"
bash Miniforge3-$(uname)-$(uname -m).sh

conda install -n base conda-libmamba-solver
conda config --set solver libmamba
conda activate base

Cloned repo succesfully with the following steps:
git clone https://github.com/ucb-bar/chipyard.git
cd chipyard

But the ./build-setup.sh riscv-tools command gives following error. Please clarify.

**
Downloading and Extracting Packages: ...working... done
Preparing transaction: ...working... done
Verifying transaction: ...working... done
Executing transaction: ...working... done
To activate this environment, use
$ conda activate /home/dgangana/rocketchip/chipyard/.conda-lock-env
To deactivate an active environment, use
 $ conda deactivate

Channels:
 - conda-forge
Platform: linux-64
Collecting package metadata (repodata.json): ...working... done
Solving environment: ...working... failed

LibMambaUnsatisfiableError: Encountered problems while solving:
  - package conda-lock-2.5.7-pyhd8ed1ab_0 requires urllib3 >=1.26.5,<2.0, but none of the providers can be installed

Could not solve for environment specs
The following packages are incompatible
├─ conda-lock 2.5.7**  is installable and it requires
│  └─ urllib3 >=1.26.5,<2.0 , which can be installed;
└─ urllib3 2.2.1.*  is not installable because it conflicts with any installable versions previously reported.

build-setup.sh: Build script failed with exit code 1 at step 1: Conda environment setup
(base) dgangana@Eva:~/rocketchip/chipyard$

gituserdeepika, 2024-08-06:
Thank you. [New issue](https://github.com/chipsalliance/rocket-chip/issues/3663) opened for chipyard error, closing this for now.

closed 2024-08-06 by gituserdeepika