chipsalliance/rocket-chip #3663: ./build-setup.sh errors in chipyard flow

gituserdeepika, 2024-08-06:
For building chipyard based rocket-chip, I followed the instructions from https://chipyard.readthedocs.io/en/latest/Chipyard-Basics/Initial-Repo-Setup.html#initial-repository-setup.

Installed Conda with the following steps:
curl -L -O "https://github.com/conda-forge/miniforge/releases/latest/download/Miniforge3-$(uname)-$(uname -m).sh"
bash Miniforge3-$(uname)-$(uname -m).sh

conda install -n base conda-libmamba-solver
conda config --set solver libmamba
conda activate base

Cloned repo succesfully with the following steps:
git clone https://github.com/ucb-bar/chipyard.git
cd chipyard

But the ./build-setup.sh riscv-tools command gives following error. Please clarify.

**
Downloading and Extracting Packages: ...working... done
Preparing transaction: ...working... done
Verifying transaction: ...working... done
Executing transaction: ...working... done
To activate this environment, use
$ conda activate /home/dgangana/rocketchip/chipyard/.conda-lock-env
To deactivate an active environment, use
$ conda deactivate

Channels:

conda-forge
Platform: linux-64
Collecting package metadata (repodata.json): ...working... done
Solving environment: ...working... failed
LibMambaUnsatisfiableError: Encountered problems while solving:

package conda-lock-2.5.7-pyhd8ed1ab_0 requires urllib3 >=1.26.5,<2.0, but none of the providers can be installed
Could not solve for environment specs
The following packages are incompatible
├─ conda-lock 2.5.7** is installable and it requires
│ └─ urllib3 >=1.26.5,<2.0 , which can be installed;
└─ urllib3 2.2.1.* is not installable because it conflicts with any installable versions previously reported.

build-setup.sh: Build script failed with exit code 1 at step 1: Conda environment setup
(base) dgangana@Eva:~/rocketchip/chipyard$

abejgonzalez, 2024-08-06:
https://github.com/ucb-bar/chipyard/pull/1970 might have fixed this issue. I would try on a newer version of Chipyard.

gituserdeepika, 2024-08-06:
I was able to resolve the urllib3 issue by updating to the correct version with the below command.
conda install -c conda-forge urllib3=1.26.7

And I am on the latest version of chipyard as shown below.
(base) dgangana@Eva:~/rocketchip/chipyard$ git pull origin main
From https://github.com/ucb-bar/chipyard
 * branch              main       -> FETCH_HEAD
Already up to date.

However I still see the following error with the './build-setup.sh riscv-tools' command. Please clarify.

**build-setup.sh error**
========== BEGINNING STEP 9: Pre-compiling FireMarshal buildroot sources ==========
To check on progress, either call marshal with '-v' or see the live output at:
/home/dgangana/rocketchip/chipyard/software/firemarshal/logs/br-base-build-2024-08-06--01-19-09-Q5LANAMRF3HFTNIF.log
.  /home/dgangana/rocketchip/chipyard/software/firemarshal/boards/firechip/base-workloads/br-base/host-init.sh
.  /home/dgangana/rocketchip/chipyard/software/firemarshal/images/firechip/br.7251/br.7251.img
Attempting to download cached image: https://raw.githubusercontent.com/firesim/firemarshal-public-br-images/main/images/firechip/br.7251/br.7251.img.zip
Unzipping cached image: /home/dgangana/rocketchip/chipyard/software/firemarshal/boards/firechip/distros/br/br.7251.img.zip
Skipping full buildroot build. Using cached image /home/dgangana/rocketchip/chipyard/software/firemarshal/images/firechip/br.7251/br.7251.img from /home/dgangana/rocketchip/chipyard/software/firemarshal/boards/firechip/distros/br/br.7251.img.zip
.  build_busybox
.  /home/dgangana/rocketchip/chipyard/software/firemarshal/images/firechip/br-base/br-base-bin
.  calc_br-base_dep
.  /home/dgangana/rocketchip/chipyard/software/firemarshal/images/firechip/br-base/br-base.img
TaskError - taskid:/home/dgangana/rocketchip/chipyard/software/firemarshal/images/firechip/br-base/br-base.img
PythonAction Error
Traceback (most recent call last):
  File "/home/dgangana/rocketchip/chipyard/.conda-env/lib/python3.10/site-packages/doit/action.py", line 461, in execute
    returned_value = self.py_callable(*self.args, **kwargs)
  File "/home/dgangana/rocketchip/chipyard/software/firemarshal/wlutil/build.py", line 603, in makeImage
    wlutil.applyOverlay(config['img'], config['overlay'])
  File "/home/dgangana/rocketchip/chipyard/software/firemarshal/wlutil/wlutil.py", line 752, in applyOverlay
    copyImgFiles(img, flist, 'in')
  File "/home/dgangana/rocketchip/chipyard/software/firemarshal/wlutil/wlutil.py", line 701, in copyImgFiles
    with mountImg(img, getOpt('mount-dir')):
  File "/home/dgangana/rocketchip/chipyard/.conda-env/lib/python3.10/contextlib.py", line 135, in __enter__
    return next(self.gen)
  File "/home/dgangana/rocketchip/chipyard/software/firemarshal/wlutil/wlutil.py", line 598, in mountImg
    assert ret == 1, f"{mntPath} already mounted. Somethings wrong"
AssertionError: /home/dgangana/rocketchip/chipyard/software/firemarshal/disk-mount already mounted. Somethings wrong

ERROR: Failed to build workload br-base.json
Log available at: /home/dgangana/rocketchip/chipyard/software/firemarshal/logs/br-base-build-2024-08-06--01-19-09-Q5LANAMRF3HFTNIF.log
ERROR: FAILURE: 1 builds failed
build-setup.sh: Build script failed with exit code 1 at step 9: Pre-compiling FireMarshal buildroot sources

abejgonzalez, 2024-08-06:
You can skip the FireMarshal setup (our SW build tool) if you are just running Rocket Chip builds. See the `--help` options (`--skip-marshal`).

gituserdeepika, 2024-08-07:
Thank you, By skipping the Firemarshal setup the build went through successfully.

closed 2024-08-07 by gituserdeepika