chipsalliance/rocket-chip #3666: Unable to generate vcd file from a riscv assembly file

gituserdeepika, 2024-08-09:
I am trying to simulate a riscv assembly file and view the waveform, but seeing the below errors, please clarify.
**Assembly test**
.section .text
  .globl _start

_start:
  li t1, 5            
  li t2, 3            

  add t3, t1, t2      # t3 = t1 + t2

  li a7, 10           
  ecall              

**Commands used to created .riscv** 
Compiling assembly test
	riscv64-unknown-elf-as -o addasm.o addasm.s
Linking
	riscv64-unknown-elf-ld -o addasm addasm.o
Creating riscv	
	riscv64-unknown-elf-objcopy -O binary addasm addasm.riscv
Checking riscv
	riscv64-unknown-elf-objdump -D -b binary -m riscv addasm.riscv

**Command used to simulate and generate vcd file**
> make run-binary-debug BINARY=$RISCV/../../sims/verilator/asm-programs/addasm.riscv
Running with RISCV=/home/dgangana/rocketchip/chipyard/.conda-env/riscv-tools
if [ "/home/dgangana/rocketchip/chipyard/.conda-env/riscv-tools/../../sims/verilator/asm-programs/addasm.riscv" != "none" ] && [ ! -f "/home/dgangana/rocketchip/chipyard/.conda-env/riscv-tools/../../sims/verilator/asm-programs/addasm.riscv" ]; then printf "\n\nBinary /home/dgangana/rocketchip/chipyard/.conda-env/riscv-tools/../../sims/verilator/asm-programs/addasm.riscv not found\n\n"; exit 1; fi
if [ "/home/dgangana/rocketchip/chipyard/.conda-env/riscv-tools/../../sims/verilator/asm-programs/addasm.riscv" != "none" ]; then riscv64-unknown-elf-objdump -D -S /home/dgangana/rocketchip/chipyard/.conda-env/riscv-tools/../../sims/verilator/asm-programs/addasm.riscv > /home/dgangana/rocketchip/chipyard/sims/verilator/output/chipyard.harness.TestHarness.RocketConfig/addasm.dump ; fi
riscv64-unknown-elf-objdump: /home/dgangana/rocketchip/chipyard/.conda-env/riscv-tools/../../sims/verilator/asm-programs/addasm.riscv: file format not recognized
make: *** [/home/dgangana/rocketchip/chipyard/common.mk:344: /home/dgangana/rocketchip/chipyard/.conda-env/riscv-tools/../../sims/verilator/asm-programs/addasm.riscv.run.debug] Error 1

jerryz123, 2024-08-09:
The file you are passing to chipyard via the BINARY flag must be a ELF binary. Not disassembly.

gituserdeepika, 2024-08-09:
Can you please provide the commands to create the ELF binary?
I am unable to find the same in https://chipyard.readthedocs.io/en/latest/index.html

jerryz123, 2024-08-09:
https://chipyard.readthedocs.io/en/latest/Software/Baremetal.html

gituserdeepika, 2024-08-09:
https://chipyard.readthedocs.io/en/latest/Software/Baremetal.html provides commands to create a binary from c file using riscv64-unknown-elf-gcc tool. Will the same tool and commands work for generating an ELF binary from an asm file?
Please note I am able to simulate binaries generated from c files, I am facing issues simulating binaries generated from asm files.
Please clarify.

gituserdeepika, 2024-08-12:
I am trying to analyze if I can stream data from memory to the Rocket CPU. But by using a C based test there is a lot of assembly code that is created due to which I am not able to do this analysis. So I prefer to write my tests natively in assembly right now for my analysis. I do not want stack etc. for now, just want to write an assembly based test, assemble the same and make Rocket execute only that. Can someone provide the steps for the same?

gituserdeepika, 2024-08-12:
Closing this for now

closed 2024-08-12 by gituserdeepika