chipsalliance/rocket-chip #3703: make verilog error

abendezu10, 2024-12-22:
[image]
Not sure whats wrong, I installed the proper dependencies but always come across this problem. I am not sure if im using the wrong versions or something.

Cameudis, 2025-02-25:
Met these error too. Solved by explicitly choosing a mill version (0.11.13) by `echo 0.11.13 > .mill-version`.

JackySu, 2025-02-26:
> Met these error too. Solved by explicitly choosing a mill version (0.11.13) by `echo 0.11.13 > .mill-version`.

Appreciate for your help! though I get more errors after that, did you manage to compile it? What's the branch/commit you checkout? Much thanks.
```
kvm:~/rocket-chip> make verilog
cd /home/ss/rocket-chip && mill emulator[freechips.rocketchip.system.TestHarness,freechips.rocketchip.system.DefaultConfig].mfccompiler.compile
[69/285] foreign-modules.dependencies.chisel.build.plugin[2.13.12].compileResour[76/285] foreign-modules.dependencies.chisel.build.plugin[2.13.12].sources.super[102/285] foreign-modules.dependencies.chisel.build.plugin[2.13.12].zincReportCa[285/285] emulator[freechips.rocketchip.system.TestHarness,freechips.rocketchip.system.DefaultConfig].mfccompiler.compile 
1 targets failed
emulator[freechips.rocketchip.system.TestHarness,freechips.rocketchip.system.DefaultConfig].mfccompiler.compile java.io.IOException: Cannot run program "firtool" (in directory "/home/ss/rocket-chip/out/emulator/freechips.rocketchip.system.TestHarness/freechips.rocketchip.system.DefaultConfig/mfccompiler/compile.dest"): error=2, no such file or directory
    java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1143)
    java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1073)
    os.proc.proc$lzycompute$1(ProcessOps.scala:138)
    os.proc.proc$1(ProcessOps.scala:137)
    os.proc.spawn(ProcessOps.scala:144)
    os.proc.call(ProcessOps.scala:87)
    millbuild.build$Emulator$mfccompiler$.$anonfun$compile$6(build.sc:229)
java.io.IOException: error=2, no such file or directory
    java.base/java.lang.ProcessImpl.forkAndExec(Native Method)
    java.base/java.lang.ProcessImpl.<init>(ProcessImpl.java:314)
    java.base/java.lang.ProcessImpl.start(ProcessImpl.java:244)
    java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1110)
    java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1073)
    os.proc.proc$lzycompute$1(ProcessOps.scala:138)
    os.proc.proc$1(ProcessOps.scala:137)
    os.proc.spawn(ProcessOps.scala:144)
    os.proc.call(ProcessOps.scala:87)
    millbuild.build$Emulator$mfccompiler$.$anonfun$compile$6(build.sc:229)
```

helloworld-star, 2025-02-27:
It seems that firtool is not installed. You can download the binary from [release 1.40.0](https://github.com/llvm/circt/releases/tag/firtool-1.40.0). It has been tested and can run successfully.

If you are using Nix, it suggests choosing version 1.56.0.