chipsalliance/rocket-chip #3722: Chisel 7 Support - ChiselAnnotation going away

tymcauley, 2025-03-06:
With https://github.com/chipsalliance/chisel/pull/4717, Chisel has removed the `ChiselAnnotation` API, which breaks a number of spots throughout rocket-chip:

- https://github.com/chipsalliance/rocket-chip/blob/ac5192c76646569ebd47d247b33cd69d6f01ce0c/src/main/scala/util/ElaborationArtefactAnnotation.scala#L41-L49
- https://github.com/chipsalliance/rocket-chip/blob/ac5192c76646569ebd47d247b33cd69d6f01ce0c/src/main/scala/rocket/DCache.scala#L26-L31
- Many uses within https://github.com/chipsalliance/rocket-chip/blob/ac5192c76646569ebd47d247b33cd69d6f01ce0c/src/main/scala/util/Annotations.scala

Can we replace the custom `DCache` `InlineInstance` trait with either `chisel3.util.experimental.InlineInstance` (in Chisel 6.6.0) or [`InlineInstanceAllowDedup`](https://github.com/chipsalliance/chisel/blob/05c87bca07ebaa611b9e4486a7e041c59ff4c628/src/main/scala/chisel3/util/experimental/Inline.scala#L45-L50) (in Chisel 7)?

Do the other annotations have any consumer, or would they be harmless to remove?

jerryz123, 2025-03-06:
It looks like ElaborationArtefactAnnotation is also unused? I think all the unused Annotations can be removed.

InlineInstance is useful, but I agree that should be changed to `experimental.InlineInstance`.

tymcauley, 2025-03-06:
> It looks like ElaborationArtefactAnnotation is also unused? I think all the unused Annotations can be removed.
> 
> InlineInstance is useful, but I agree that should be changed to `experimental.InlineInstance`.

Regarding DCache's `InlineInstance`, should we try to use `chisel3.util.experimental.InlineInstanceAllowDedup`, since that doesn't block dedup, like `chisel3.util.experimental.InlineInstance` does?

jerryz123, 2025-03-06:
Yeah, good point. It would be unintuitive for inlined instances to block dedup

tymcauley, 2025-03-07:
I'll submit a PR to get rid of most of the annotations except for DCache's `InlineInstance`. That PR will be Chisel-6 compatible. Then I'll make another which replaces DCache's `InlineInstance`

jackkoenig, 2025-03-07:
I want to clarify that, while Chisel 7 removes `ChiselAnnotation`, it's still possible to annotate things, you just have to go directly to `FirrtlAnnotation`[^1]. That being said, since it's basically impossible to implement user-defined annotations in CIRCT, there's a lot less utility in them than there once was anyway--it's best to use upstream annotations where possible.

[^1]: Admittedly this API is not released but will be in 6.7.0. I can prioritize getting 6.7.0 out if it helps y'all at all.

tymcauley, 2025-03-07:
> I want to clarify that, while Chisel 7 removes `ChiselAnnotation`, it's still possible to annotate things, you just have to go directly to `FirrtlAnnotation`[1](#user-content-fn-1-a3f223b2745d92b1488adc93a6564054). That being said, since it's basically impossible to implement user-defined annotations in CIRCT, there's a lot less utility in them than there once was anyway--it's best to use upstream annotations where possible.
> ## Footnotes
> 
>     1. Admittedly this API is not released but will be in 6.7.0. I can prioritize getting 6.7.0 out if it helps y'all at all. [↩](#user-content-fnref-1-a3f223b2745d92b1488adc93a6564054)

Ya, that might help us resolve https://github.com/ucb-bar/constellation/issues/81 and https://github.com/chipsalliance/diplomacy/issues/18

tymcauley, 2025-03-07:
> I want to clarify that, while Chisel 7 removes `ChiselAnnotation`, it's still possible to annotate things, you just have to go directly to [...]

@jackkoenig I don't see the `FirrtlAnnotation` API in Chisel right now, all I see are calls to `chisel3.experimental.annotate(...)`.

jackkoenig, 2025-03-07:
I've tagged v6.7.0, the artifacts should be on Maven in ~2 hours: https://github.com/chipsalliance/chisel/releases/tag/v6.7.0

@tymcauley the new `chisel3.experimental.annotate` takes `=> Seq[firrtl.annotations.Annotation]` as its 2nd argument list, for example[^1]:
```scala
  def addResource(blackBoxResource: String): Unit = {
    chisel3.experimental.annotate(self)(Seq(BlackBoxInlineAnno.fromResource(blackBoxResource, self.toNamed)))
  }
```

Also see the PR (https://github.com/chipsalliance/chisel/pull/4643) that changed from the old to new for an example.

[^1]: https://github.com/chipsalliance/chisel/blob/425bdd2bd6132de684d0fdebe5f5a09aca61acb5/src/main/scala/chisel3/util/BlackBoxUtils.scala#L47

tymcauley, 2025-03-08:
Yup, we can use the new annotation API from https://github.com/chipsalliance/chisel/pull/4643 to fix the remaining `ChiselAnnotation` in `DCache`, I'll submit a PR for that.

tymcauley, 2025-03-10:
Resolved by #3724

PR #3724 (2025-03-10): Update to Chisel 6.7, use new annotation API
Resolves #3722.

**Related issue**: #3722

**Type of change**: other enhancement

**Impact**: no functional change

**Development Phase**: implementation

**Release Notes**

Update to Chisel 6.7.0
closed 2025-03-10 by tymcauley